fossasia/neurolab-hardware #46: PCB Layout for communication circuitry

CloudyPadmal, 2018-11-24:
Components placement and routing for Bluetooth, WiFi and UART

Niharika248, 2019-03-28:
For WiFi and Bluetooth routing on PCB Layout of NeuroLab which WiFi Module will good to use ESP32 or ESP8266 ?
ESP32 has got in built Bluetooth 4.2 and BLE and UART pins 2 but for ESP8266 there is no Bluetooth in built
Component placement and circuitry on the neurolab board mightwill be easy with ESP32

Niharika248, 2019-03-28:
will the **firmware** be written on to any EPROM or ROM?
How can we manage that on board?
i think an SD card driver slot would be good if we have on board

Niharika248, 2019-04-04:
**_For Routing*_* - ESP32 module with built in antenna and  and an RS232, SPI interface would be good.
 ESP32 can interface with other systems to provide Wi-Fi and Bluetooth functionality through its SPI / SDIO or I2C / UART interfaces.

**_ For Firmware*_* - The data saved in the flash memory remains there even when the ESP32 resets or when power is removed. The flash memory is very similar to the EEPROM. Both are non-volatile memories.

[image]