fossasia/neurolab-hardware #55: Find alternative IC for AD7173

CloudyPadmal, 2018-11-29:
[AD7173](https://www.mouser.sg/ProductDetail/Analog-Devices/AD7173-8BCPZ?qs=sGAEpiMZZMsrChSOYEGTCegWOOYfy6BbjXESgxq25zw%3d) IC is the heart of the Neuro Lab device. But this IC costs nearly 17 USD. Do some research on finding a cheaper version with similar or close capacities to this IC and evaluate the possibility of a replacement.

wijjj, 2019-01-08:
Hey Padmal, you can also go ahead and directly ask the brainduino developers. :)

Niharika248, 2019-03-27:
AD7173-8 series IC can be replaced with AD7172-4 which is of same AD717X series and has got similar features to that of AD7173 IC the only difference is it has True Rail to Rail buffers. Price - AD7172-4 IC costs approx $7.8 USD which is almost 10 times lesser than the existing IC

OPA2140 has got SMD packaging structure and has similar capacities to OPA2111 IC.
****

CloudyPadmal, 2019-03-28:
Hi, thank you very much for the findings.
- AD ICs seems to have $1 difference and AD7173 has 16 channels while AD7172 has only 8 channels. But this selection would be good and lets take it into consideration.
- OPA2140 is a good choice :+1:

Niharika248, 2019-03-28:
Thats great
thank you
i will check up on furthur issues

tharaka27, 2019-04-03:
Can't we generalize what we expect from this IC? So it will be easy for anyone who is searching for an alternative IC.

1. Analog to Digital converter
2. 24bit resolution
3. 16 channel 
4. Sigma delta architecture 

is this all?

CloudyPadmal, 2019-04-03:
Hi, We are expecting following features;
- Price
- Resolution (Need to check the optimum bits needed to have quality signal. This can be achieved either by amplifying the signal to a higher voltage, quantize with lower resolution ADC chip or not putting much effort on amplifying stage and use a high resolution ADC chip)
- Dimensions and package of IC
- Power consumption and levels
- Number of channels (8 preferable)

arishuynhvan, 2019-04-03:
@CloudyPadmal 
[AD7770](https://www.mouser.sg/ProductDetail/Analog-Devices/AD7770ACPZ?qs=sGAEpiMZZMvTvDTV69d2Qi%2FLvSr2sd1YD5UCctgE8hA%3D) and [AD7779](https://www.mouser.sg/ProductDetail/Analog-Devices/AD7779ACPZ?qs=sGAEpiMZZMvTvDTV69d2QoIMHehgkbmswLCUTk0UP3g%3D) seem like possible alternatives given the key features mentioned.
Please check out the comparison table in this [link](https://docs.google.com/document/d/1zKS3nxPLIu-V_WlWRN0DV4loRpERMbo-w6ZKGOsefvM/edit?usp=sharing) 

May I also know if the inbuilt digital filters (sinc1,3,5) of AD7173 will be used?